AnHardt/Marlin #84: Visualisation of x-sled-rotation

AnHardt, 2018-09-05:
```scad
// x_rotation.scad

// Animated demonstration for nozzle
// and probe seeing not the same height
// caused by rototion of the x-sled,
// around the x-axis,
// when moving in x-dirextion.
// 
// Load in OpenScad.
// Start animation with about 10 FPS
// and 50 steps.
// Observe from all sides, then take
// a look directly along the x-axis.
// See nozle (red) and probe (blue)
// not flying at the same height.
// Play with the next 3 values.

z_till = 10; // misalignment of the z-towers in °.
// when 0 machine is perfectly adjusted.

nozzle_dist = 20; // from center of rotaton in mm.
// When 0 the effect to the nozzles height is minimized.

probe_dist = 40; // from center of rotaton in mm.
// When same as nozzle_dist the probe will see the same height as the nozzle.
// probe_dist > nozzle_dist --> overcompensation
// probe_dist < nozzle_dist --> undercompensation

rod_len = 300;
rod_dia = 8;
x_rod_dist2 = 25;
cone_size = 10;

module rod(){
  translate([0,0,-rod_len/2])
  cylinder(d= rod_dia, h= rod_len);    
}

module z_rod(){
  rod();
}

module x_rod() {
  rotate([0,90,0]) 
  rod();    
};

module nozzle() {
  color("red")
  translate([0,nozzle_dist,0])
  cylinder(d1=0, d2=cone_size, h=cone_size);
}
module probe() {
  color("blue")
  translate([0,probe_dist,0])
  cylinder(d1 = 0, d2 = cone_size, h = cone_size);
}

rotate([-z_till,0,0]) translate([-rod_len/2,0,0]) z_rod();
rotate([z_till,0,0]) translate([+rod_len/2,0,0]) z_rod();

x_end_shift = x_rod_dist2 * tan(z_till);
echo("x_end_shift",x_end_shift);
x_rot = atan2(x_end_shift, rod_len/2);
echo("x_rot",x_rot);

rotate([0,0,x_rot]) translate([0,0,-x_rod_dist2]) x_rod();
rotate([0,0,-x_rot]) translate([0,0,+x_rod_dist2]) x_rod();

x = $t * rod_len - rod_len/2;
x_sled_rot = 2*z_till * $t - z_till; 
echo("x_sled_rot",x_sled_rot);
rotate([x_sled_rot,0,0]) translate([x,0,0]) nozzle();
rotate([x_sled_rot,0,0]) translate([x,0,0]) probe();
```

emaayan, 2018-12-13:
thanks, i've uploaded a youtube video showing your animation along with links to the bugs that i've found 
https://www.youtube.com/watch?v=rPU4ZAPJTNc

habitoti, 2021-03-31:
This is a bit of an older post, but I also had that issue and came up with a tiny tool to compute an appropriate compensation. [Checkout this blog post](http://rombos.de/wordpress/2021/03/31/bltouch-frustration/).

closed 2024-02-12 by AnHardt